nlef/moonraker-telegram-bot #207: add a new eta parameter "estimate" / "middle value" / "average"

Staubgeborener, 2022-10-14:
### What would you like to see added?

Right now the only possible eta values are `slicer` or `file`:

```shell
[telegram_ui]
#eta_source: slicer
#	You can choose, which value to use for remaining time estimation.
#	Default value is slicer.
#	Allowed values: slicer, file
```

Klipper can calculate a third eta value based on the middle value of `filament` and `file`. In my case, this is one of the most presicely eta calculation i can get. While `slicer` eta time shows me a value of `0:05:35`, the average value of `file` (`0:09:44`) and `filament` (`0:03:20`) give me a eta value of `0:06:32` which is in fact accurate to the second.

Maybe it's a opportunity to add a third value `filament` to the `moonraker-telegram-bot` and let the user decide which numbers should be combined and determine the average value from this.

aka13-404, 2022-10-18:
Thank you for writing such a detailed feature request, it's so refreshing :)
What I am wondering though, I have not had time to check, are you sure it's a klipper thing? 
The slicer and filament variables are pulled from moonraker, we do not touch them ourselves. 

Is this "middle" thing a new addition which we missed? If it is, adding it should be easy. If it is not, I am not sure if we want to add it. We want to keep the amount of calculations/things we do ourselves to a necessary minimum, because the bot is not an active component of the ecosystem, but rather a displaying/controlling entity.

Staubgeborener, 2022-10-18:
Incant tell you if this calculation is from klipper or moonraker. Maybe it's a mainsail related feature? I can't tell you that's In my setup, I am using klipper with mainsail. In mainsail you can go to settings and there is a section for eta. Mainsail ask on which base the eta should be calculated and you can choose between different sources (file, filament, slicer). At exactly this step you can also choose more than one source to calculate an average value.

CODeRUS, 2022-11-03:
see: https://github.com/mainsail-crew/mainsail/blob/develop/src/store/printer/getters.ts#L1013