nlef/moonraker-telegram-bot #213: Bot is not deleting timelapses after sending is taking up a lot of space[Bug]: A useful, descriptive title

Orangeking05, 2022-10-25:
### What happened and in what context?

its not deleting the timelaples after sending them

### If needed, paste the relevant bot log contents here.

[Uploading telegram.log…]()

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

Orangeking05, 2022-11-13:
> 

Attached (in the wrong spot sorry)

aka13-404, 2022-11-13:
You try to take way to many photos (at least every 5 seconds, most likely about every 2 seconds) and at the same time connectivity issues. 
At the same time you make the bot spam every 5(most likely about every 3) seconds a picture towards telegram.
All sort of funny behaviour is gonna happen if you do that.
Your connectivity issues are happening, propably because Telegram is banning you, since you spam a trillion images per second for you status updates. 

Make sure you have a reasonable config, before reporting issues. Readjust your config, make sure you are not having connectivity problems, and everything sohould work as expected.

Orangeking05, 2022-11-14:
[telegram-2.conf.TXT](https://github.com/nlef/moonraker-telegram-bot/files/10000247/telegram-2.conf.TXT)
Is this better? Also is there any way i can do the timelapse by height with my first layer being 0.28 tall and all subsecent layer 0.2? Thank you for you Help

aka13-404, 2022-11-14:
Yes, this looks better. 
It does take a picture, even at 0.28, since it is >0.2.

dhrstrik, 2022-11-14:
and from my other printer .
#  Please refer to the wiki(https://github.com/nlef/moonraker-telegram-bot/wiki) for detailed information on how to configure the bot

[bot]
server: localhost
bot_token: 5617696554:AAGkgL9JETVGBeklfCapjHUwzI0OOXB6SN0
chat_id: 952761771

[camera]
host: http://localhost:8080/?action=stream

[progress_notification]
percent: 5
height: 5
time: 5

[timelapse]
cleanup: true
height: 0.2
time: 5
target_fps: 30

closed 2022-11-14 by aka13-404