nlef/moonraker-telegram-bot #227: Docker image

moookino, 2022-12-08:
### What would you like to see added?

Can you add docker image, which could be used as part of https://github.com/mkuf/prind stack?
I've created feature request to add telegram bot https://github.com/mkuf/prind/issues/45

Thanks in advance

mkuf, 2022-12-12:
I just tried to add this Image to Prind in https://github.com/mkuf/prind/pull/48

It seems, the Image is missing Dependencies.

```
Traceback (most recent call last):
  File "/opt/moonraker-telegram-bot/bot/main.py", line 41, in <module>
    from websocket_helper import WebSocketHelper
  File "/opt/moonraker-telegram-bot/bot/websocket_helper.py", line 11, in <module>
    from notifications import Notifier
  File "/opt/moonraker-telegram-bot/bot/notifications.py", line 11, in <module>
    from camera import Camera
  File "/opt/moonraker-telegram-bot/bot/camera.py", line 16, in <module>
    import cv2  # type: ignore
ModuleNotFoundError: No module named 'cv2'
```

This might be resolvable by adding the package to the requirements file instead of installing it via apt. 

-Markus

nlef, 2022-12-12:
@nlef I fix Dockerfile. My mistake with system packages links in venv

nlef, 2022-12-12:
@mkuf I fixed Dockerfile. My mistake with system packages links in venv

mkuf, 2022-12-13:
Created a PR to add support for arm/v7 and arm64/v8: https://github.com/nlef/moonraker-telegram-bot/pull/231

nlef, 2023-01-03:
@mkuf @moookino 
build pipeline is now fixed and docker images for arm are available at [docker hub](https://hub.docker.com/repository/docker/lefskiy/moonraker-telegram-bot)
Tag `latest` would be available with the next release.

mkuf, 2023-01-03:
Great 💯 
The Service has been added to prind v1.4.0 yesterday. https://github.com/mkuf/prind/releases/tag/v1.4.0  
Prind is currently using the `development` tag, it may move to `latest` in a future release as soon as the tag is available.

closed 2023-02-05 by aka13-404