nlef/moonraker-telegram-bot #233: Adding multiple chatid’s to the bot whitelist.

nc1914, 2022-12-20:
### What would you like to see added?

Hi, I have a problem where the bot can only message one person at a time. Every time I try to add another person to the whitelist, the bot breaks and kicks them off and kicks everyone out. Is there any way to allow multiple people to get status updates and message the bot please?

Thanks

nc1914, 2022-12-20:
Just occurred to me to clarify, I am adding the chatids into the telegram.conf file

aka13-404, 2022-12-20:
You can't do that as you have noticed. You can add the bot to a group though, where it will send status message updates, but can not be controlled from. 
Multiple people controlling the bot is not planned in the foreseeable future, since this calls for permissions, logging, conflicts etc.

nc1914, 2022-12-20:
Thanks for the fast response! Do I simply just add my bot to a group then and it will post updates there as well or is there some amount of configuring to do first please?

closed 2022-12-27 by aka13-404

nc1914, 2022-12-31:
> Not quite, you have to enter the group id in config. Check the wiki for the parameter.
> 
> #233

Sorry, But I can’t find the word group inthe wiki anywhere, and I can’t just add the groupID as it reports as an error in the config.

nc1914, 2022-12-31:
Also, sorry for the delay, I had to reinstall klipper from scratch a couple of times due to a machine-breaking error.

aka13-404, 2022-12-31:
https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config

nc1914, 2022-12-31:
Oh… I am sorry I didn’t see this before. I just need to add the config lines onto the end of the config file, right? Thanks so much for your patience.