nlef/moonraker-telegram-bot #237: [Bug]: Timelapse compilation error

hardl0ck, 2023-01-03:
### What happened and in what context?

The bot works great, sends pictures, videos, but never compiles timelapses.
The main printer works with the uStreamer, I tried to do a clean install with the mjpg-srteamer - the same thing.
The camera is raspicam 5MP.

Raspbain Bullseye (Linux 5.15.84-v7+ #1610 SMP Mon Dec 19 18:47:48 GMT 2022 armv7l GNU/Linux)

### If needed, paste the relevant bot log contents here.

Current Moonraker telegram bot config
[bot]
server = localhost:7125
power_device = printer
light_device = led_light
upload_path = /home/pi/printer_data/gcodes

[telegram_ui]
progress_update_message = true
silent_progress = true

[status_message_content]
content = progress, height, filament_length, print_duration, finish_time
heaters = extruder, heater_bed
fans = rpi_fan
sensors = mcu_temp
moonraker_devices = light, dryer

[camera]
host = http://localhost/webcam/?action=stream
flip_vertically = false
flip_horizontally = false
video_duration = 15
fps = 10

[progress_notification]
percent = 1
height = 1
time = 60

[timelapse]
basedir = ~/moonraker-telegram-bot-timelapse
copy_finished_timelapse_dir = /home/pi/timelapse/t_bot
cleanup = false
height = 0.2
time = 0
target_fps = 15
last_frame_duration = 1
send_finished_lapse = true

*******************************************************************
2023-01-03 13:53:23,011 - __main__ - ERROR - 

2023-01-03 13:53:24,325 - websocket_helper - ERROR - 'estimated_time'
2023-01-03 13:55:20,451 - __main__ - ERROR - Job e769bec8025c4a9baf14690f5803091f raised[Errno 2] No such file or directory: '/home/pi/moonraker-telegram-bot-timelapse/test 400mm-s.gcode_1970-01-01_01-00/lapse.lock'
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/timelapse.py", line 263, in _send_lapse
    ) = self._camera.create_timelapse(lapse_filename, gcode_name, info_mess)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 420, in create_timelapse
    return self._create_timelapse(printing_filename, gcode_name, info_mess)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 455, in _create_timelapse
    lock_file.touch()
  File "/usr/lib/python3.9/pathlib.py", line 1304, in touch
    fd = self._raw_open(flags, mode)
  File "/usr/lib/python3.9/pathlib.py", line 1116, in _raw_open
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/moonraker-telegram-bot-timelapse/test 400mm-s.gcode_1970-01-01_01-00/lapse.lock'
2023-01-03 13:56:07,860 - klippy - INFO - 'filename' has the same value as the current: 

### If needed, paste the relevant bot journal contents here.

-- Journal begins at Thu 2022-09-22 01:23:40 BST, ends at Tue 2023-01-03 14:00:16 GMT. --
Jan 03 13:56:08 sapphire python[4510]: 2023-01-03 13:56:08,072 - apscheduler.executors.default - INFO - Job "Notifier._send_message (trigger: date[2023-01-03 13:56:07 GMT], next run at: 2023-01-03 13:56:07 GMT)" executed successfully
Jan 03 13:56:07 sapphire python[4510]: 2023-01-03 13:56:07,868 - apscheduler.executors.default - INFO - Running job "Notifier._send_message (trigger: date[2023-01-03 13:56:07 GMT], next run at: 2023-01-03 13:56:07 GMT)" (scheduled at 2023-01-03 13:56:07.862323+00:00)
Jan 03 13:56:07 sapphire python[4510]: 2023-01-03 13:56:07,867 - apscheduler.scheduler - INFO - Removed job 656aee3cb78b4cb6b47b5a191106f123
Jan 03 13:56:07 sapphire python[4510]: 2023-01-03 13:56:07,864 - apscheduler.scheduler - INFO - Added job "Notifier._send_message" to job store "default"
Jan 03 13:56:07 sapphire python[4510]: 2023-01-03 13:56:07,860 - klippy - INFO - 'filename' has the same value as the current:
Jan 03 13:55:20 sapphire python[4510]: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/moonraker-telegram-bot-timelapse/test 400mm-s.gcode_1970-01-01_01-00/lapse.lock'
Jan 03 13:55:20 sapphire python[4510]:     return self._accessor.open(self, flags, mode)
Jan 03 13:55:20 sapphire python[4510]:   File "/usr/lib/python3.9/pathlib.py", line 1116, in _raw_open
Jan 03 13:55:20 sapphire python[4510]:     fd = self._raw_open(flags, mode)
Jan 03 13:55:20 sapphire python[4510]:   File "/usr/lib/python3.9/pathlib.py", line 1304, in touch
Jan 03 13:55:20 sapphire python[4510]:     lock_file.touch()
Jan 03 13:55:20 sapphire python[4510]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 455, in _create_timelapse
Jan 03 13:55:20 sapphire python[4510]:     return self._create_timelapse(printing_filename, gcode_name, info_mess)
Jan 03 13:55:20 sapphire python[4510]:   File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 420, in create_timelapse
Jan 03 13:55:20 sapphire python[4510]:     ) = self._camera.create_timelapse(lapse_filename, gcode_name, info_mess)
Jan 03 13:55:20 sapphire python[4510]:   File "/home/pi/moonraker-telegram-bot/bot/timelapse.py", line 263, in _send_lapse
Jan 03 13:55:20 sapphire python[4510]:     retval = job.func(*job.args, **job.kwargs)
Jan 03 13:55:20 sapphire python[4510]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
Jan 03 13:55:20 sapphire python[4510]: Traceback (most recent call last):
Jan 03 13:55:20 sapphire python[4510]: 2023-01-03 13:55:20,451 - __main__ - ERROR - Job e769bec8025c4a9baf14690f5803091f raised[Errno 2] No such file or directory: '/home/pi/moonraker-telegram-bot-timelapse/test 400mm-s.gcode_1970-01-01_01-00/lapse.lock'
Jan 03 13:55:17 sapphire python[4510]: 2023-01-03 13:55:17,980 - apscheduler.executors.default - INFO - Job "Notifier._send_print_finish (trigger: date[2023-01-03 13:55:15 GMT], next run at: 2023-01-03 13:55:15 GMT)" executed successfully
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,241 - apscheduler.scheduler - INFO - Removed job 4d52e60c36604355a281607506179f11
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,232 - apscheduler.executors.default - INFO - Running job "Notifier._send_print_finish (trigger: date[2023-01-03 13:55:15 GMT], next run at: 2023-01-03 13:55:15 GMT)" (scheduled at 2023-01-03 13:55:15.210393+00:00)
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,226 - apscheduler.scheduler - INFO - Removed job e769bec8025c4a9baf14690f5803091f
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,220 - apscheduler.executors.default - INFO - Running job "Timelapse._send_lapse (trigger: date[2023-01-03 13:55:15 GMT], next run at: 2023-01-03 13:55:15 GMT)" (scheduled at 2023-01-03 13:55:15.206877+00:00)
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,212 - apscheduler.scheduler - INFO - Added job "Notifier._send_print_finish" to job store "default"
Jan 03 13:55:15 sapphire python[4510]: 2023-01-03 13:55:15,209 - apscheduler.scheduler - INFO - Added job "Timelapse._send_lapse" to job store "default"
Jan 03 13:53:24 sapphire python[4510]: 2023-01-03 13:53:24,325 - websocket_helper - ERROR - 'estimated_time'
Jan 03 13:53:24 sapphire python[4510]: 2023-01-03 13:53:24,324 - websocket - ERROR - error from callback <bound method WebSocketHelper.websocket_to_message of <websocket_helper.WebSocketHelper object at 0x63b96238>>: 'estimated_time'
Jan 03 13:53:24 sapphire python[4510]: 2023-01-03 13:53:24,272 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Jan 03 13:53:24 sapphire python[4510]: 2023-01-03 13:53:24,244 - websocket - WARNING - websocket connected
Jan 03 13:53:24 sapphire python[4510]: 2023-01-03 13:53:24,231 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Jan 03 13:53:23 sapphire python[4510]: 2023-01-03 13:53:23,432 - apscheduler.scheduler - INFO - Scheduler started
Jan 03 13:53:23 sapphire python[4510]: 2023-01-03 13:53:23,038 - apscheduler.scheduler - INFO - Scheduler started
Jan 03 13:53:23 sapphire python[4510]: 2023-01-03 13:53:23,011 - __main__ - ERROR -
Jan 03 13:53:16 sapphire systemd[1]: Started Moonraker Telegram Bot SV1.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

hardl0ck, 2023-01-05:
update. with MJPG-streamer - timelapse working, 
but with uStreamer - not worked. Folder with screenshots are not created.
i`m try tomorow to reinstall all on clean system with uStreamer.

aka13-404, 2023-02-01:
Can not reproduce, timelapse does not depend on where the input is form. Reopen if you have steps to reliably reproduce from a clean installation. We use ustreamer all the time for testing without problems.

closed 2023-02-01 by aka13-404