nlef/moonraker-telegram-bot #240: [problem]  some data not working

boiydiego777, 2023-01-10:
### What happened and in what context?

1. update link moonraker to printer_data folder location
2.problem is in telegram data is not working progress staying at 0% , filament staying at 0m , printing time 0:00:00 , estimated time is ok finish time also ok, picture also working..

Kind regards
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/10380672/telegram.log)
[journal.txt](https://github.com/nlef/moonraker-telegram-bot/files/10380673/journal.txt)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

boiydiego777, 2023-01-11:
log file and journal attached , somebody can help?

nlef, 2023-01-11:
@boiydiego777 
judging by the attached logs, you have problems with the Internet connection.

closed 2023-02-01 by aka13-404