nlef/moonraker-telegram-bot #241: [Bug]: Unsure how to get status messages working after update.

dviera421, 2023-01-11:
### What happened and in what context?

I am no longer getting progress notifications, however, I still receive a Print Job Started message and Printer Status Change message. I think I broke it from updating through the Mainsail GUI and am unclear of how to fix it.  I know there was an announcement but I am still unclear of how-to fix after updating and if I am doing something else wrong. Can you please elaborate the process to fix after updating? Thanks. 

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2023-01-11:
No log - no issue.

dviera421, 2023-01-11:
I am not sure how to generate a telegram log. All I can find is klippy log, moonraker log and crowsnest log. Thanks anyways.

aka13-404, 2023-01-11:
I'd say your safest bet is to reinstall, if there is no log. 
Also, why are you running development, are you testing something specific?

dviera421, 2023-01-11:
I will try that. I believe it defaulted to development when I was filling out the ticket, my mistake.

dviera421, 2023-01-23:
Unfortunately I wasn’t able to figure it out. Currently have it uninstalled ☹   From: Klaas ReinekeSent: Monday, January 23, 2023 9:57 AMTo: nlef/moonraker-telegram-botCc: dviera421; MentionSubject: Re: [nlef/moonraker-telegram-bot] [Bug]: Unsure how to get status messages working after update. (Issue #241) same issue here, did an update today, messaging not working. Tried to setup a secrets.conf restarted everything and still get no messages. Any directions how to solve it @dviera421 ?

closed 2023-02-01 by aka13-404