nlef/moonraker-telegram-bot #242: Sending arbitrary files by command

aka13-404, 2023-01-11:
### What would you like to see added?

For some workflows (like testing resonances, or logging stuff to a file) it may be useful to be able to attach a file, video or a picture.

This should be a MESSAGE_RESPOND type command, which in turn has a path parameter and errors out, if the file attached is too big. Path validation is done by the user.

nlef, 2023-01-11:
Commands examples:
```
RESPOND PREFIX=tg_send_image MSG="path='../imgs/nosignal.png' message='asd message' "
RESPOND PREFIX=tg_send_video MSG="path='../imgs/nosignal.png' message='asd message' "
RESPOND PREFIX=tg_send_document MSG="path='../imgs/nosignal.png' message='asd message' "
```
`path` parameteris required, `message` is optional

aka13-404, 2023-01-15:
Sending a file which is too large should result in the same error as happens with timelapse, otherwise very nice

nlef, 2023-01-29:
Extended syntax added for sending image albums\file groups. Example
```
RESPOND PREFIX=tg_send_image MSG="path=['../imgs/nosignal.png', '../imgs/nopreview.png', '../imgs/nosignal.png'] message='lala message' "
RESPOND PREFIX=tg_send_video MSG="path=['../imgs/nosignal.png', '../imgs/nopreview.png', '../imgs/nosignal.png'] message='lala message' "
RESPOND PREFIX=tg_send_document MSG="path=['../imgs/nosignal.png', '../imgs/nopreview.png', '../imgs/nosignal.png'] message='lala message' "
```
the old syntax is still available
```
RESPOND PREFIX=tg_send_image MSG="path='../imgs/nosignal.png' message='asd message' "
RESPOND PREFIX=tg_send_video MSG="path='../imgs/nosignal.png' message='asd message' "
RESPOND PREFIX=tg_send_document MSG="path='../imgs/nosignal.png' message='asd message' "
```

aka13-404, 2023-02-01:
Synthax is not conforming with #191, needs to be unified.

closed 2023-02-05 by aka13-404