nlef/moonraker-telegram-bot #245: [Bug]: Clutter in moonraker-telegram-bot-timelapse Folder fills up Diskspace

its-Blackpoint, 2023-01-13:
### What happened and in what context?

On 2 of my Printers that use this Plugin I Suddenly had 0.1kb Diskspace and Couldnt Upload anything to it. Turns out all of the Media the Telegram Bot is Crating is stored in the "moonraker-telegram-bot-timelapse" Folder and won't be deleted. 

Would be nice to have an Auto Purge or at least an Option to Delete the Files via the Mainsail UI and not have to do this via ssh

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2023-01-13:
No log - no issue

its-Blackpoint, 2023-01-13:
Pls read my Issue. How the Heaven sould i log this?

its-Blackpoint, 2023-01-13:
So i'll try my best to describe the Issue i hope thets ok even if i still dont know how to log this

Before i Delete the content in "moonraker-telegram-bot-timelapse" Folder it looks like this 
[image]

And in Mainsail i have a Free Space of 198MB

[image]

then i Delete the Content in the Folder

[image]
03de6.png)

after i have more then 800MB free Space! This altrough my ls-lh on hte folder said it only contains 1.5MB

![after delete mainsail](https://user-images.githubusercontent.com/57731174/212428712-1732b202-ac17-4645-be0a-c651601

Something here is very disk consuming

aka13-404, 2023-01-13:
I am interested in your configuration, and latest prints. If you for example configure the bot not to delete images after building, and then say "it's not deleting", that's not a bug. If there are errors with filesystem access or similar - it's an issue.

Please attach logs.

its-Blackpoint, 2023-01-13:
*******************************************************************
Current Moonraker telegram bot config
[bot]
server = localhost

[camera]
host = http://localhost:8080/?action=stream

[progress_notification]
percent = 5
height = 5
time = 5

[timelapse]
cleanup = true
height = 0.2
time = 5
target_fps = 30

*******************************************************************
2022-12-31 17:56:19,500 - __main__ - ERROR - 

2022-12-31 17:56:19,511 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1070, in <module>
    bot_updater = start_bot(configWrap.secrets.token, configWrap.bot.socks_proxy)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 977, in start_bot
    request_kwargs=request_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/updater.py", line 290, in __init__
    else arbitrary_callback_data
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 105, in __init__
    private_key_password=private_key_password,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 194, in __init__
    self.token = self._validate_token(token)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 358, in _validate_token
    raise InvalidToken()
telegram.error.InvalidToken: Invalid token
Stack (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 65, in handle_exception
    stack_info=True,

*******************************************************************
Current Moonraker telegram bot config
[bot]
server = localhost

[camera]
host = http://localhost:8080/?action=stream

[progress_notification]
percent = 5
height = 5
time = 5

[timelapse]
cleanup = true
height = 0.2
time = 5
target_fps = 30

*******************************************************************
2022-12-31 17:56:28,993 - __main__ - ERROR - 

2022-12-31 17:56:29,004 - __main__ - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 1070, in <module>
    bot_updater = start_bot(configWrap.secrets.token, configWrap.bot.socks_proxy)
  File "/home/pi/moonraker-telegram-bot/bot/main.py", line 977, in start_bot
    request_kwargs=request_kwargs,
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/updater.py", line 290, in __init__
[... 2970 lines cut ...]
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/camera.py", line 416, in take_lapse_photo
    outfile.write(photo.getvalue())
OSError: [Errno 28] No space left on device
2023-01-13 21:40:32,720 - timelapse - ERROR - [Errno 28] No space left on device
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thre
*******************************************************************
Current Moonraker telegram bot config
[bot]
server = localhost

[camera]
host = http://localhost:8080/?action=stream

[progress_notification]
percent = 5
height = 5
time = 5

[timelapse]
cleanup = true
height = 0.2
time

aka13-404, 2023-01-13:
Please attach it as a file.

closed 2023-01-14 by its-Blackpoint