nlef/moonraker-telegram-bot #248: telegram bot stopps responding from time to time

maruhe, 2023-01-25:
### What happened and in what context?

the telegram bot stopps responding from time to time
mostly while creating a timelapse
sometimes while printing
here the logs from today where it sopped updting while printing
more logs will follow with future crashes :-)

### If needed, paste the relevant bot log contents here.

*******************************************************************
Current Moonraker telegram bot config
[bot]
server = localhost

[camera]
host = http://localhost:8080/?action=stream

[progress_notification]
percent = 5
height = 0.5
time = 0

[timelapse]
manual_mode = true
cleanup = true
height = 0
time = 0
target_fps = 15
basedir = ~/moonraker-telegram-bot-timelapse
last_frame_duration = 5

[telegram_ui]
buttons = [status,pause,cancel,resume],[files,emergency,macros,shutdown],[CaseLight_0,CaseLight_1,CaseLight_5,CaseLight_10]
require_confirmation_macro = false
hidden_macros = Attach_Probe_Lock, Dock_Probe_UnlockDocks, G32, M300, PRINT_END, PRINT_START, status_busy, status_calibrating_z, status_cleaning, status_heating, status_homing, status_leveling, status_meshing, status_off, status_printing, status_ready, test_speed_fast, UPDATE_IDLE_TIMEOUT

....
teyt was to long for github
....

2023-01-25 21:46:40,954 - klippy - INFO - 'filename' has the same value as the current: 
2023-01-25 21:57:30,324 - websocket_helper - ERROR - Canceled by new editmessagemedia request
2023-01-25 21:57:41,373 - __main__ - ERROR - Job 9422b81a935c4e73bc53a8751a3a74b2 raisedCanceled by new editmessagemedia request
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/pi/moonraker-telegram-bot/bot/notifications.py", line 186, in _notify
    self._status_message.edit_media(media=InputMediaPhoto(photo))
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/message.py", line 2219, in edit_media
    return self.bot.edit_message_media(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 133, in decorator
    result = func(*args, **kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 2919, in edit_message_media
    return self._message(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/ext/extbot.py", line 202, in _message
    result = super()._message(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 339, in _message
    result = self._post(endpoint, data, timeout=timeout, api_kwargs=api_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/bot.py", line 298, in _post
    return self.request.post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 359, in post
    result = self._request_wrapper('POST', url, fields=data, **urlopen_kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
    raise BadRequest(message)
telegram.error.BadRequest: Canceled by new editmessagemedia request
2023-01-25 21:58:13,084 - __main__ - ERROR - Job 4701b7baaa6241e8a4ec06e83c1fc302 raisedCanceled by new editmessagemedia request
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/apscheduler/executors/base.py", line 125, in run_job
    retval = job.func(*job.args, **job.kwargs)
[... 143 lines cut ...]
Jan 25 22:00:35 v1pi python[735]: 2023-01-25 22:00:35,911 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Jan 25 22:00:22 v1pi python[735]: 2023-01-25 22:00:22,260 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2023-01-25 22:00:19 CET], next run at: 2023-01-25 22:00:19 C>
Jan 25 22:00:19 v1pi python[735]: 2023-01-25 22:00:19,725 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2023-01-25 22:00:19 CET], next run at: 2023-01-25 22>
Jan 25 22:00:19 v1pi python[735]: 2023-01-25 22:00:19,725 - apscheduler.scheduler - INFO - Removed job 70236d7ee3444185bdca04ad93b91bcb
Jan 25 22:00:19 v1pi python[735]: 2023-01-25 22:00:19,723 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Jan 25 22:00:09 v1pi python[735]: 2023-01-25 22:00:09,642 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2023-01-25 22:00:07 CET], next run at: 2023-01-25 22:00:07 C>
Jan 25 22:00:07 v1pi python[735]: 2023-01-25 22:00:07,478 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2023-01-25 22:00:07 CET], next run at: 2023-01-25 22>
Jan 25 22:00:07 v1pi python[735]: 2023-01-25 22:00:07,477 - apscheduler.scheduler - INFO - Removed job c04eaf2c2941442a89361d39dad094f7
Jan 25 22:00:07 v1pi python[735]: 2023-01-25 22:00:07,475 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Jan 25 22:00:04 v1pi python[735]: 2023-01-25 22:00:04,288 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2023-01-25 22:00:01 CET], next run at: 2023-01-25 22:00:01 C>
Jan 25 22:00:01 v1pi python[735]: 2023-01-25 22:00:01,705 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2023-01-25 22:00:01 CET], next run at: 2023-01-25 22>
Jan 25 22:00:01 v1pi python[735]: 2023-01-25 22:00:01,704 - apscheduler.scheduler - INFO - Removed job 35a5004fa8d9465388f2baded709d2a0
Jan 25 22:00:01 v1pi python[735]: 2023-01-25 22:00:01,703 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Jan 25 21:59:47 v1pi python[735]: 2023-01-25 21:59:47,517 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2023-01-25 21:59:44 CET], next run at: 2023-01-25 21:59:44 C>
Jan 25 21:59:46 v1pi python[735]: 2023-01-25 21:59:46,680 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2023-01-25 21:59:44 CET], next run at: 2023-01-25 21:59:44 C>
Jan 25 21:59:44 v1pi python[735]: 2023-01-25 21:59:44,908 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2023-01-25 21:59:44 CET], next run at: 2023-01-25 21>
Jan 25 21:59:44 v1pi python[735]: 2023-01-25 21:59:44,908 - apscheduler.scheduler - INFO - Removed job 89b62d4ba5ee43d290c2b9302ec8d139

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2023-02-01:
Please post full logs. This is not helping to pinpoint your issue at all.

maruhe, 2023-02-01:
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/10561869/telegram.log)
here a full logfile (printer is currently printing) - it should countain several crashes - also a hang while printing and one while creating timelapse in the last 3-4 days

maruhe, 2023-02-02:
clicking emergency-stop and then firmware_restart within a second seems to be a big problem for the telegram-bot also

maruhe, 2023-03-01:
here it seems to just be unable to update the telegram-message
just hope the timelapse continues as I restarted the bot
[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/10856575/telegram.log)

maruhe, 2023-03-02:
i control timlapse manual in gcode but after bot-restart no more picture was made and also no timelapse in the end

aka13-404, 2024-11-25:
Stale, closing

closed 2024-11-25 by aka13-404