nlef/moonraker-telegram-bot #253: moonraker-telegram-bot version cannot be parsed by Moonraker

JWRu, 2023-02-06:
### What happened and in what context?

The Moonraker Update Manager will show "? UNKNOWN" after an update to version v1.5-0-gb9b4700.
According to the Moonraker specialists the moonraker-telegram-bot dev tagged the latest release with a version that Moonraker can't parse.  The tags have to be compatible with the semantic version standard. 
https://semver.org/
[image]

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

JWRu, 2023-02-06:
I think it's just a typo. "v1.5-0..." should be changed to ""v1.5.0...".

aka13-404, 2023-02-06:
Thank you for being so quick!

To be honest, I am not sure where the -0 is coming from, the release is v1.5, but it seems that for semver it should be `v1.5.0`. I have corrected it, I'd appreciate if you can tell if its alright now.

JWRu, 2023-02-06:
The indication in Moonraker Update Manager didn't change. I tried a manual git pull but got the message that the package is already up to date.

aka13-404, 2023-02-06:
Well, you now how to wait, or delete everything and redownload. I hope it don't bothers you too much :(.
But we propably will have fixes very soon, when people will start using new features, and will roll over to 1.5.1 pretty fast, if that helps your bother

Arksine, 2023-02-07:
FWIW, the `-0` is generated by `git describe` and indicates the number of commits since the tag. 

I decided to relax the tag requirement in a new commit to Mooraker.  The tag can have fewer than 3 parts, so Moonraker will parse `v1.5`.  Anyone with a tag of `v1.5` should be able to update Moonraker, refresh available updates, then update `moonraker-telegram-bot`.

The only thing to keep in mind going forward is that a tag cannot have more than 3 parts, so for example a tag of `v1.5.0.1` will likely cause an issue.

JWRu, 2023-02-07:
The Moonraker update did the job. Everything is "UP-TO-DATE" now. Many thanks.

aka13-404, 2023-02-07:
> FWIW, the `-0` is generated by `git describe` and indicates the number of commits since the tag.
>
> I decided to relax the tag requirement in a new commit to Mooraker. The tag can have fewer than 3 [...]
  
You are a magician, aprreciate it. Now I have one less place to make mistakes :D
Closing this then.

closed 2023-02-07 by aka13-404

cpasoft, 2023-07-29:
I have this problem. I have updated Moonraker. Here is a screenshot of my versions.
I have already tried a complete uninstall and reinstall. I think this was after a recent moonraker update.
[image]

cpasoft, 2023-07-29:
Note:

cpasoft, 2023-07-29:
Note:
Kiauh is detecting well BTW
[image]

cpasoft, 2023-07-31:
Is there a way to reopen this??... Or I must to create a new one...

Arksine, 2023-07-31:
It looks like this could be a regression in Mainsail.

Version 2.6.0:
[image]

Version 2.6.2 (also seems to occur on 2.6.1)
[image]

Its possible that Mainsail introduced its own tag validation.

bigsubgit, 2024-01-10:
Hello everyone, is it solved this? I've "unknown" yet.