nlef/moonraker-telegram-bot #258: [FR] Add progress notification for first layer

SckyzO, 2023-03-01:
### What would you like to see added?

I suggest to add boolean functionality / feature to add status of first layer only.

[progress_notification]
percent: 10
height: 5
time: 5
show_first_layer: true/false

Often it is the first layer that fails in 3D prints, so it could allow us to follow the progress, but in case of bad 1st layer we can stop the print

aka13-404, 2023-03-01:
Stupid question, does the runtime g-code not work for this? I havent tested this scenario explicitely, but it should work, afaik.

SckyzO, 2023-03-02:
Is not a stupid question, I will try this gcode. But  I think it would be a good idea to just simplify this functionality. Either way, I think it would be useful.