nlef/moonraker-telegram-bot #273: Message whem Filament runs out

Dave2526, 2023-05-19:
Hello,
Thanks for this very good addon, but i have the following Problem: 
Today i have started a Print and the Filament was broken. So my filement Sensor Pause the Print, but i got No Message via the bot. Ist this a settting i am Missing or ist this Not possible at the moment?

Thanks!

Otann, 2023-05-23:
You can add the `RESPOND` command to your filament runout config, for example:

```
[filament_motion_sensor runout]
detection_length: 14.0
extruder: extruder
pause_on_runout: FALSE
runout_gcode:
  {action_respond_info("RUNOUT: Filament runout")}
  RESPOND PREFIX=tgnotify MSG="Filament runout detected"
  PAUSE
```

Dave2526, 2023-05-24:
Ok right I have searched for a runout code..  The respond  command was not in my mind, thanks s lot

closed 2023-05-24 by Dave2526