nlef/moonraker-telegram-bot #282: [Bug]: Red Overlayer from Telegram

SpaceRichard, 2023-07-22:
### What happened and in what context?

The timelaps video gets a red overlay at the end. 
The 5sec video don't and also the klipper timelapse not, so I would locate this issue to the timelaps generation within telegram.

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

komandrik, 2023-07-24:
I have the same problem. The bot sends normal pictures to Telegram, but renders them in red.

https://github.com/nlef/moonraker-telegram-bot/assets/40995038/b8ea3140-a5af-4538-bfd7-f8e592d96b23

[image]

komandrik, 2023-08-01:
I didn't wait for a response, reinstalled debian11 x32, and the video stopped showing in red.

SpaceRichard, 2023-08-06:
Really wonder what is the reason. A fix would be great instead of reinstalling the system.

Darthyson, 2023-08-06:
Maybe an outdated version of OpenCV. You can try `pip3 install opencv-python`. At least it worked for me today on a raspberry 4 with bullseye x64.

SpaceRichard, 2023-08-07:
Thanks @Darthyson
I  also run `pip3 install --upgrade pip` as only OpenCV was not solving the issues. 
Now the red layer is gone.

I'm not sure if the project can or should do something in order to prevent this, so I keep it open. Let the author decide.

horego, 2023-08-18:
Same issue for me. A proper fix would help.

aka13-404, 2024-11-25:
With 2.0.0 the stuff around opencv got a rework + default is now not to use it alltogether. Please reopen if issues of that sort persist.

closed 2024-11-25 by aka13-404