nlef/moonraker-telegram-bot #283: Multi-person management functionality

berkovskiy, 2023-08-03:
### What would you like to see added?

Hello. I know I was [Issues](https://github.com/nlef/moonraker-telegram-bot/issues/233) and you answered the question clearly and clearly. But still, this functionality is not enough. I would like to manage the printer together with a colleague. Maybe I can provide any assistance ?

nlef, 2024-01-06:
Management requests from multiple users were discussed in #233 
At the moment I have no plans to implement this.
If there are problems with sending messages to channels or groups, open a separate ticket describing the problem and attach the bot log.

aka13-404, 2024-11-25:
This could slowly move on in releases following 2.0