nlef/moonraker-telegram-bot #284: [Bug]: AGAIN... moonraker-telegram-bot version cannot be parsed by Moonraker (or Mainsaill)

cpasoft, 2023-08-04:
### What happened and in what context?

The Moonraker Update Manager will show "? UNKNOWN" after an update to version v1.5-0-gb9b4703.
Arksine has said that It looks like this could be a regression in Mainsail.

[image]

Related to:
https://github.com/nlef/moonraker-telegram-bot/issues/253

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

mkryuchkov, 2023-12-04:
Any updates? May be workarounds?

Phil1988, 2023-12-22:
Seems not yet updated...

aka13-404, 2024-11-25:
Sorry, a bit stale, and can not reproduce. Please reopen if issue persists in 2.0.0

closed 2024-11-25 by aka13-404