nlef/moonraker-telegram-bot #285: Add "enabled" option to timelaps config

mrodyhin, 2023-08-12:
### What would you like to see added?

Now it's possible to enable/disable timelaps via runtime gcode commands, ([see that example](https://github.com/nlef/moonraker-telegram-bot/wiki/Interacting-with-klipper#runtime-lapse-and-notification-setting)), but it's not possible to set initial enabled state in .conf file (it's always enabled). Would be useful to make it configurable.

GrimmSimon, 2023-08-19:
Maybe you can try this:
```
[timelapse]
...
manual_mode: true
```
the default value is False.

or you comment out the complete `[timelaps]` section.

I hope this helps you.

nlef, 2024-01-06:
@mrodyhin 
Parameter `enabled` is not present in the semantics of klipper configurations. Therefore, I do not plan to add this option at this time.
To disable the functionality, just comment out the corresponding section of the configuration.

closed 2024-01-06 by nlef