nlef/moonraker-telegram-bot #289: [Bug]: secrets file is not being picked up?

Otann, 2023-10-16:
### What happened and in what context?

I have the telegram.conf file

```
#  Please refer to the wiki(https://github.com/nlef/moonraker-telegram-bot/wiki) for detailed information on how to configure the bot

[bot]
server: localhost
# bot_token: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
# chat_id: 1

[secrets]
secrets_path: /home/pi/printer_data/config

[progress_notification]
time: 600
percent: 10

[status_message_content]
content: progress, eta, finish_time, m117_status, last_update_time
sensors: rpi, canbus_mcu, chamber
heaters: extruder, heater_bed
fans: chamber, nevermore

[camera]
host: http://localhost:8080/?action=stream
rotate: 180
fps: 30

[telegram_ui]
buttons: [status,pause,cancel,resume],[emergency,macros,shutdown],[HEATSOAK, COOLDOWN]
require_confirmation_macro: false
progress_update_message: false
status_message_m117_update: true
```

and I have `secrets.conf` file in the same folder next to it:

```
[secrets] 
bot_token: xxxxx
chat_id: 111111
```

But when the bot starts it complains that the values are not provided. And if I use fake values, the log says that the token is invalid.

It looks like the secrets file is not being used, or am I doing something wrong?

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

tinyfluffs, 2024-01-30:
Same issue happens here, clean install from KIAUH.

vasyl83, 2024-02-24:
just tried and having the same issue

nlef, 2024-03-08:
@Otann @vasyl83 @tinyfluffs 
For now you can specify the full path with the name of the secrets file.
Default secrets filename fixed in [6601fdf](https://github.com/nlef/moonraker-telegram-bot/commit/6601fdfee262e426c411c1b8bd308ea22d2b6c1c)

aka13-404, 2024-11-25:
Fixed in 2.0.0

closed 2024-11-25 by aka13-404