nlef/moonraker-telegram-bot #290: [Bug]: tgcustom_keyboard not working

napalmz, 2023-10-25:
### What happened and in what context?

I created a macro to send a message with buttons but each time it is called, nothing happens. Only an error is produced in the logs.

This is the macro:
```
[gcode_macro _TG_SHUTDOWN_IMMINENT]
gcode:
    {% set bot_msg = "message='Shutdown in 15 min.', buttons=[{name='Stop', command='SHUTDOWN_ABORT'},{name='Cancel', command='delete'}]" %}
    RESPOND PREFIX=tgcustom_keyboard MSG="{bot_msg}"
```

### If needed, paste the relevant bot log contents here.

2023-10-25 13:15:25,397 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f8e340> is not JSON serializable
2023-10-25 13:17:17,493 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f99740> is not JSON serializable
2023-10-25 13:22:21,660 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f996c0> is not JSON serializable
2023-10-25 13:24:57,296 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f8e3c0> is not JSON serializable
2023-10-25 13:27:03,812 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f997c0> is not JSON serializable

### If needed, paste the relevant bot journal contents here.

-- Journal begins at Wed 2023-10-25 10:17:04 CEST, ends at Wed 2023-10-25 13:33:56 CEST. --
Oct 25 13:28:53 raspberrypi2 python[2062]: 2023-10-25 13:28:53,499 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Oct 25 13:28:53 raspberrypi2 python[2062]: 2023-10-25 13:28:53,483 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:28:53 raspberrypi2 python[2062]: 2023-10-25 13:28:53,480 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:28:55 CEST)" executed successfully
Oct 25 13:28:53 raspberrypi2 python[2062]: 2023-10-25 13:28:53,479 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:28:55 CEST)" (scheduled at 2023-10-25 13:28:53.478134+02:00)
Oct 25 13:28:51 raspberrypi2 python[2062]: 2023-10-25 13:28:51,478 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:28:51 raspberrypi2 python[2062]: 2023-10-25 13:28:51,466 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:28:51 raspberrypi2 python[2062]: 2023-10-25 13:28:51,462 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:28:53 CEST)" executed successfully
Oct 25 13:28:51 raspberrypi2 python[2062]: 2023-10-25 13:28:51,462 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:28:53 CEST)" (scheduled at 2023-10-25 13:28:51.461164+02:00)
Oct 25 13:28:49 raspberrypi2 python[2062]: 2023-10-25 13:28:49,461 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:28:49 raspberrypi2 python[2062]: 2023-10-25 13:28:49,460 - klippy - ERROR - Klippy Host not connected
Oct 25 13:28:49 raspberrypi2 python[2062]: 2023-10-25 13:28:49,438 - websocket_helper - WARNING - klippy disconnect detected with message: notify_klippy_disconnected
Oct 25 13:27:03 raspberrypi2 python[2062]: 2023-10-25 13:27:03,812 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f997c0> is not JSON serializable
Oct 25 13:27:03 raspberrypi2 python[2062]: 2023-10-25 13:27:03,811 - websocket - ERROR - error from callback <bound method WebSocketHelper.websocket_to_message of <websocket_helper.WebSocketHelper object at 0x7f812a4520>>: <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f997c0> is not JSON serializable
Oct 25 13:26:58 raspberrypi2 python[2062]: 2023-10-25 13:26:58,458 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Oct 25 13:26:58 raspberrypi2 python[2062]: 2023-10-25 13:26:58,440 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:26:58 raspberrypi2 python[2062]: 2023-10-25 13:26:58,429 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:27:00 CEST)" executed successfully
Oct 25 13:26:58 raspberrypi2 python[2062]: 2023-10-25 13:26:58,428 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:27:00 CEST)" (scheduled at 2023-10-25 13:26:58.427098+02:00)
Oct 25 13:26:56 raspberrypi2 python[2062]: 2023-10-25 13:26:56,427 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:26:56 raspberrypi2 python[2062]: 2023-10-25 13:26:56,401 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:26:56 raspberrypi2 python[2062]: 2023-10-25 13:26:56,398 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:26:58 CEST)" executed successfully
Oct 25 13:26:56 raspberrypi2 python[2062]: 2023-10-25 13:26:56,397 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:26:58 CEST)" (scheduled at 2023-10-25 13:26:56.396634+02:00)
Oct 25 13:26:54 raspberrypi2 python[2062]: 2023-10-25 13:26:54,397 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:26:54 raspberrypi2 python[2062]: 2023-10-25 13:26:54,395 - klippy - ERROR - Klippy Host not connected
Oct 25 13:26:54 raspberrypi2 python[2062]: 2023-10-25 13:26:54,380 - websocket_helper - WARNING - klippy disconnect detected with message: notify_klippy_disconnected
Oct 25 13:24:57 raspberrypi2 python[2062]: 2023-10-25 13:24:57,296 - websocket_helper - ERROR - <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f8e3c0> is not JSON serializable
Oct 25 13:24:57 raspberrypi2 python[2062]: 2023-10-25 13:24:57,296 - websocket - ERROR - error from callback <bound method WebSocketHelper.websocket_to_message of <websocket_helper.WebSocketHelper object at 0x7f812a4520>>: <telegram.inline.inlinekeyboardbutton.InlineKeyboardButton object at 0x7f70f8e3c0> is not JSON serializable
Oct 25 13:24:50 raspberrypi2 python[2062]: 2023-10-25 13:24:50,889 - apscheduler.scheduler - INFO - Removed job ws_reschedule
Oct 25 13:24:50 raspberrypi2 python[2062]: 2023-10-25 13:24:50,875 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:24:50 raspberrypi2 python[2062]: 2023-10-25 13:24:50,865 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:24:52 CEST)" executed successfully
Oct 25 13:24:50 raspberrypi2 python[2062]: 2023-10-25 13:24:50,864 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:24:52 CEST)" (scheduled at 2023-10-25 13:24:50.862863+02:00)
Oct 25 13:24:48 raspberrypi2 python[2062]: 2023-10-25 13:24:48,863 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:24:48 raspberrypi2 python[2062]: 2023-10-25 13:24:48,850 - websocket_helper - WARNING - Error received from websocket: {'code': -32601, 'message': 'Method not found'}
Oct 25 13:24:48 raspberrypi2 python[2062]: 2023-10-25 13:24:48,846 - apscheduler.executors.default - INFO - Job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:24:50 CEST)" executed successfully
Oct 25 13:24:48 raspberrypi2 python[2062]: 2023-10-25 13:24:48,845 - apscheduler.executors.default - INFO - Running job "WebSocketHelper.reshedule (trigger: interval[0:00:02], next run at: 2023-10-25 13:24:50 CEST)" (scheduled at 2023-10-25 13:24:48.845030+02:00)
Oct 25 13:24:46 raspberrypi2 python[2062]: 2023-10-25 13:24:46,845 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
[... cut ...]
Oct 25 13:15:13 raspberrypi2 python[2062]: 2023-10-25 13:15:13,276 - apscheduler.scheduler - INFO - Added job "WebSocketHelper.reshedule" to job store "default"
Oct 25 13:15:13 raspberrypi2 python[2062]: 2023-10-25 13:15:13,274 - klippy - ERROR - Klippy Host not connected
Oct 25 13:15:13 raspberrypi2 python[2062]: 2023-10-25 13:15:13,253 - websocket_helper - WARNING - klippy disconnect detected with message: notify_klippy_disconnected

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

Wildecity, 2023-12-14:
Have you found any solution to this? I'm having the same issue

nlef, 2024-06-07:
commit 11b6893 should fix this issue
@Wildecity could you test it?

aka13-404, 2024-11-25:
Works on latest release, closing.

closed 2024-11-25 by aka13-404