nlef/moonraker-telegram-bot #293: [Bug]: Incredible high frame rate on timelapses

Phil1988, 2023-11-21:
### What happened and in what context?

First:
Thanks for this great software - its really handy and great!

However I found a bug that hopefully could be eliminated.
Installed the latest version like 4 days ago.
In the config I have set the telegram bot to take snapshots every 10 sec.

```
[timelapse]
send_finished_lapse: true
cleanup: true
time: 10
target_fps: 20
min_lapse_duration: 5
max_lapse_duration: 10
last_frame_duration: 2
```

I just finished a 20h print and it generated a 12 second video with a size of 90MB and 719.01 frames/second.

I expected the video to cut all the frames in such a manner, that it gets close to 20 fps.

What did I do wrong or can you check if the creation of the video can be improved to meet the target fps?
(I know a snapshot every 10s is much on a 20h print, but sometimes I also print things in 7min and I would like not to edit the settings for every print)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

Phil1988, 2024-11-28:
@aka13-404 
I get this message:
```
Printer onlinemoonraker-telegram-bot: v1.5-0-gb9b47003

Unknown/bad items in section [timelapse]:
  limit_fps: 30

Please correct the configuration according to the wiki (https://github.com/nlef/moonraker-telegram-bot/wiki)
```

At the update manager I see that its up-to-date while you said v2 is released:
[image]

How to update properly?

Phil1988, 2024-12-07:
Can someone please reopen this, it as it seems to be unsolved for me?

nlef, 2025-01-19:
@Phil1988
pls update the bot ot version 2.1 and and check the wiki for a change in the description of the `limit_fps` parameter

Phil1988, 2025-01-20:
@nlef 
Thanks!
I just updated and changed it to `limit_fps: true` to get it dropping the frames.
In my opinion this should be the default (not `false`) or do you see any benefit in a higher frame rate?

I havent tested an further but I can confirm that the error messages are gone now 👍 

I'll reopen, once I tested it and get problems.

Thanks for your work mate!

closed 2025-01-20 by Phil1988

Phil1988, 2025-01-21:
Thanks for your explainaition - even if it escalated a bit 😄
My prints range from 2min-20h, print heights from a few mm to 30cm.

Taking pictures based on z-height will not work/look ok for small objects, so it must be time based.
If you have a better idea how to do it, I am happy to hear:
```
[timelapse]
basedir: ~/moonraker-telegram-bot-timelapse
send_finished_lapse: true
cleanup: true
time: 20
target_fps: 20
limit_fps: true
min_lapse_duration: 5
max_lapse_duration: 10
last_frame_duration: 2
```