nlef/moonraker-telegram-bot #297: [Bug]: ERROR:  Could not build wheels for ujson which use PEP 517 and cannot be installed directly

Nocentinia, 2024-01-25:
### What happened and in what context?

I can't install the bot, compare this error
ERROR: Could not build wheels for ujson which use PEP 517 and cannot be installed directly

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2024-04-07:
@Nocentinia 
#313 should help you

RunyAlfaNew, 2024-05-27:
I have a similar error:
"Could not build wheels for  wsaccel, wich is required to intsall pyproject.tom-based projects"

I have a fresh installation of Moonraker on Ubuntu 24

brusarp, 2024-06-10:
> I have a similar error: "Could not build wheels for wsaccel, wich is required to intsall pyproject.tom-based projects"
> 
> I have a fresh installation of Moonraker on Ubuntu 24

Yea, me too... Same error...

AlanAndre, 2024-07-29:
> > I have a similar error: "Could not build wheels for wsaccel, wich is required to intsall pyproject.tom-based projects"
> > I have a fresh installation of Moonraker on Ubuntu 24
> 
> Yea, me too... Same error...

For me the fix was updating `wsaccel==0.6.6` in `moonraker-telegram-bot/scripts/requirements.txt`
Using venv with Python 3.12.3.

Tilitili, 2024-08-15:
> For me the fix was updating `wsaccel==0.6.6` in `moonraker-telegram-bot/scripts/requirements.txt` Using venv with Python 3.12.3.

Thank you. Everything works

aka13-404, 2024-11-25:
This should be resolved in latest version. Reopen if not so.

closed 2024-11-25 by aka13-404