nlef/moonraker-telegram-bot #301: Custom printer name setup in config

Hoeby, 2024-02-18:
### What would you like to see added?

When the printer is started, the first message is "printer online".

I have multi printers, all with there own klipper environment. But also with there own hardware. Both give the first message "printer online"

Would like to modify the name "printer" so that i know which one is online.

Tried in mainsail to modify the printer name, but that isn't send in the telegram messages

Altblechkutscher, 2024-02-19:
Don't you use individual bots? In my case, I have named the bot corresponding to my printer, which gives me a nice notification:

 [image]

Hoeby, 2024-02-19:
No, i use 1 bot for all. It is also used for my home automation. This gives me all information in 1 bot view.

I know that i can't use the commands from telegram, to control the printer. Because more environments are lissening to the same bot. But i think that has nothing to do with the name "printer online"

Altblechkutscher, 2024-02-20:
So if I've understood correctly, something like this?

[image]

[image]

[image]

Hoeby, 2024-02-20:
Yes, that is exactly the idea.

If i see it correctly, it is waiting for approve, to get merged?

Hoeby, 2024-03-07:
Be aware that sending commands by telegram doesn't work then. If you are adware for this and don't use it, then it is no problem

aka13-404, 2024-11-25:
We do not support and will never support multiple bots per instance. This is insanity, do you want to run one klipper/moonraker instance for multiple printers as well? All that to not create an additional bot via botfather? :D 

If you want to aggregate notifications in a single place, you can use a group, as suggested above.
Progress on control from a group will be tracked in #283. 

@Altblechkutscher You are welcome to open a PR request, this is a nice fun little option, if you think it has value. On our side nothing speaks

closed 2024-11-25 by aka13-404

Altblechkutscher, 2024-11-25:
@aka13-404 Indeed I did, which you can see here  #https://github.com/nlef/moonraker-telegram-bot/pull/302