nlef/moonraker-telegram-bot #304: [Bug]: Progress notification not working

zvarnes, 2024-03-02:
### What happened and in what context?

From the vanilla install I'm getting no progress notifications. It's currently set to '50'. 
With this set I would expect to receive a notification when my print is at %50 and %100 according to the documentation.

### If needed, paste the relevant bot log contents here.

[telegram.log](https://github.com/nlef/moonraker-telegram-bot/files/14468049/telegram.log)

### If needed, paste the relevant bot journal contents here.

pi@octopi:~ $ sudo journalctl -r -u moonraker-telegram-bot
-- Logs begin at Fri 2024-03-01 21:29:52 EST, end at Fri 2024-03-01 23:10:45 EST. --
Mar 01 23:02:00 octopi python[716]: 2024-03-01 23:02:00,589 - apscheduler.executors.default - INFO - Job "Notifier._send_print_start_info (trigger: date[2024-03-01 23:0Mar 01 23:01:59 octopi python[716]: 2024-03-01 23:01:59,102 - apscheduler.scheduler - INFO - Removed job c0620a76054c4c0b847e17b428ff9217
Mar 01 23:01:59 octopi python[716]: 2024-03-01 23:01:59,095 - apscheduler.executors.default - INFO - Running job "Notifier._send_print_start_info (trigger: date[2024-03Mar 01 23:01:59 octopi python[716]: 2024-03-01 23:01:59,093 - apscheduler.scheduler - INFO - Added job "Notifier._send_print_start_info" to job store "default"
Mar 01 22:38:56 octopi python[716]: 2024-03-01 22:38:56,054 - apscheduler.executors.default - INFO - Job "Notifier._send_print_finish (trigger: date[2024-03-01 22:38:49Mar 01 22:38:55 octopi python[716]: telegram.error.BadRequest: Canceled by new editmessagemedia request
Mar 01 22:38:55 octopi python[716]:     raise BadRequest(message)
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 279, in _request_wrapper
Mar 01 22:38:55 octopi python[716]:     result = self._request_wrapper('POST', url, fields=data, **urlopen_kwargs)
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 359, in post
Mar 01 22:38:55 octopi python[716]:     f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 300, in _post
Mar 01 22:38:55 octopi python[716]:     result = self._post(endpoint, data, timeout=timeout, api_kwargs=api_kwargs)
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 344, in _message
Mar 01 22:38:55 octopi python[716]:     message_thread_id=message_thread_id,
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 213, in _message
Mar 01 22:38:55 octopi python[716]:     api_kwargs=api_kwargs,
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 3007, in edit_message_media
Mar 01 22:38:55 octopi python[716]:     result = func(*args, **kwargs)
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 134, in decorator
Mar 01 22:38:55 octopi python[716]:     inline_message_id=None,
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/message.py", line 2334, in edit_media
Mar 01 22:38:55 octopi python[716]:     self._status_message.edit_media(media=InputMediaPhoto(photo))
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot/bot/notifications.py", line 189, in _notify
Mar 01 22:38:55 octopi python[716]:     retval = job.func(*job.args, **job.kwargs)
Mar 01 22:38:55 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/apscheduler/executors/base.py", line 125, in run_job
Mar 01 22:38:55 octopi python[716]: Traceback (most recent call last):
Mar 01 22:38:55 octopi python[716]: 2024-03-01 22:38:55,106 - __main__ - ERROR - Job c84c75509e194272b1f6e3bd4bea8ac8 raisedCanceled by new editmessagemedia request
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,648 - apscheduler.scheduler - INFO - Removed job 2e0f723b63b14f85be46aefccd1ac9b4
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,647 - apscheduler.executors.default - INFO - Running job "Notifier._send_print_finish (trigger: date[2024-03-01 
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,644 - apscheduler.scheduler - INFO - Removed job 09ac179688db4e4aa8f58a264bd5eb88
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,643 - apscheduler.executors.default - INFO - Job "Timelapse._send_lapse (trigger: date[2024-03-01 22:38:49 EST],Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,643 - apscheduler.executors.default - INFO - Running job "Timelapse._send_lapse (trigger: date[2024-03-01 22:38:Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,641 - apscheduler.scheduler - INFO - Removed job c84c75509e194272b1f6e3bd4bea8ac8
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,640 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2024-03-01 22:38:49 ESMar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,638 - apscheduler.scheduler - INFO - Added job "Notifier._send_print_finish" to job store "default"
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,637 - apscheduler.scheduler - INFO - Added job "Timelapse._send_lapse" to job store "default"
Mar 01 22:38:49 octopi python[716]: 2024-03-01 22:38:49,635 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Mar 01 22:04:59 octopi python[716]: 2024-03-01 22:04:59,089 - apscheduler.executors.default - INFO - Job "Notifier._notify (trigger: date[2024-03-01 22:04:52 EST], nextMar 01 22:04:52 octopi python[716]: 2024-03-01 22:04:52,816 - apscheduler.scheduler - INFO - Removed job 3151b1aeec904ca5abda14792343b9d1
Mar 01 22:04:52 octopi python[716]: 2024-03-01 22:04:52,815 - apscheduler.executors.default - INFO - Running job "Notifier._notify (trigger: date[2024-03-01 22:04:52 ESMar 01 22:04:52 octopi python[716]: 2024-03-01 22:04:52,813 - apscheduler.scheduler - INFO - Added job "Notifier._notify" to job store "default"
Mar 01 21:34:30 octopi python[716]: telegram.error.TimedOut: Timed out
Mar 01 21:34:30 octopi python[716]:     raise TimedOut() from error
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 261, in _request_wrapper
Mar 01 21:34:30 octopi python[716]:     result = self._request_wrapper('POST', url, fields=data, **urlopen_kwargs)
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/utils/request.py", line 359, in post
Mar 01 21:34:30 octopi python[716]:     f'{self.base_url}/{endpoint}', data=data, timeout=effective_timeout
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 300, in _post
Mar 01 21:34:30 octopi python[716]:     result = self._post(endpoint, data, timeout=timeout, api_kwargs=api_kwargs)
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 344, in _message
Mar 01 21:34:30 octopi python[716]:     message_thread_id=message_thread_id,
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/ext/extbot.py", line 213, in _message
Mar 01 21:34:30 octopi python[716]:     message_thread_id=message_thread_id,
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 763, in send_photo
Mar 01 21:34:30 octopi python[716]:     result = func(*args, **kwargs)
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot-env/lib/python3.7/site-packages/telegram/bot.py", line 134, in decorator
Mar 01 21:34:30 octopi python[716]:     disable_notification=self.silent_status,
Mar 01 21:34:30 octopi python[716]:   File "/home/pi/moonraker-telegram-bot/bot/notifications.py", line 402, in _send_print_start_info
Mar 01 21:34:30 octopi python[716]:     retval = job.func(*job.args, **job.kwargs)
[... cut ...]
Mar 01 21:34:30 octopi python[716]: The above exception was the direct cause of the following exception:
Mar 01 21:34:30 octopi python[716]: telegram.vendor.ptb_urllib3.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='api.telegram.org', port=443): Read timed 
Mar 01 21:34:30 octopi python[716]:     raise exc_cls(*args)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

zvarnes, 2024-03-02:
# Update:

So it does appear to be working, but it just updates the original message. I don't get any notifications for this. Is this working as intended, or am I missing something?

# Progression: 
[image]
[image]
[image]

nlef, 2024-03-07:
@zvarnes 
`silent_progress: True` in  [telegram_ui section](https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#telegram_ui) may help you

zvarnes, 2024-03-09:
> @zvarnes `silent_progress: True` in [telegram_ui section](https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#telegram_ui) may help you also `progress_update_message` will add bzz on status updates

@nlef This solved my issue! Thank you

closed 2024-03-09 by zvarnes