nlef/moonraker-telegram-bot #310: [Bug]: "Printer online" message every 30sec or so

sochartgit, 2024-03-12:
### What happened and in what context?

I've installed the script as explained on the wiki but I am sure I've missed something.
I get a "Printer online" message on the telegram bot every 30 seconds.
Which is really annoying of course.
Could you please tell me how to disable that ?

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2024-03-15:
@sochartgit 
It looks like the bot can't start.
Please attach the bot service log

sochartgit, 2024-03-15:
The bot works, I get the other messages.
I got rid of the module. Thanks and good luck.

@sochartgit<https://github.com/sochartgit>
It looks like the bot can't start.
Please attach the bot service log

closed 2024-03-15 by nlef