nlef/moonraker-telegram-bot #318: [Bug]: Power_Devices (Eg. Light) does not work if monraker is not on port 80

yeahimongithub, 2024-07-28:
### What happened and in what context?

If the moonraker is configured with a port Eg:
```
[bot]
server: moonraker:7125
```

The light-device does not work anymore.
In the logs, you see error-messages saying, that the host 'moonraker' with port '80' is not available.

since the "server"-configuration got split up into "host" and "port", the initialiaztion of the power-devices does not work anymore:
``` python
light_power_device = PowerDevice(configWrap.bot_config.light_device_name, configWrap.bot_config.host)
```

i fixed it locally by adding the port
``` python
light_power_device = PowerDevice(configWrap.bot_config.light_device_name, configWrap.bot_config.host + ':' + str(configWrap.bot_config.port))
```

but I'm not sure if this is the intended behaviour.

### If needed, paste the relevant bot log contents here.

2024-07-28T08:20:53.092173413Z     self.endheaders()
2024-07-28T08:20:53.092186273Z   File "/usr/lib/python3.11/http/client.py", line 1277, in endheaders
2024-07-28T08:20:53.092199509Z     self._send_output(message_body, encode_chunked=encode_chunked)
2024-07-28T08:20:53.092212699Z   File "/usr/lib/python3.11/http/client.py", line 1037, in _send_output
2024-07-28T08:20:53.092226105Z     self.send(msg)
2024-07-28T08:20:53.092239021Z   File "/usr/lib/python3.11/http/client.py", line 975, in send
2024-07-28T08:20:53.092253065Z     self.connect()
2024-07-28T08:20:53.092265969Z   File "/opt/venv/lib/python3.11/site-packages/urllib3/connection.py", line 236, in connect
2024-07-28T08:20:53.092279659Z     self.sock = self._new_conn()
2024-07-28T08:20:53.092292623Z                 ^^^^^^^^^^^^^^^^
2024-07-28T08:20:53.092305576Z   File "/opt/venv/lib/python3.11/site-packages/urllib3/connection.py", line 211, in _new_conn
2024-07-28T08:20:53.092319094Z     raise NewConnectionError(
2024-07-28T08:20:53.092332300Z urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f93fff2c690>: Failed to establish a new connection: [Errno 111] Connection refused
2024-07-28T08:20:53.092347852Z 
2024-07-28T08:20:53.092360624Z The above exception was the direct cause of the following exception:
2024-07-28T08:20:53.092373975Z 
2024-07-28T08:20:53.092386762Z Traceback (most recent call last):
2024-07-28T08:20:53.092402500Z   File "/opt/venv/lib/python3.11/site-packages/requests/adapters.py", line 667, in send
2024-07-28T08:20:53.092416242Z     resp = conn.urlopen(
2024-07-28T08:20:53.092429221Z            ^^^^^^^^^^^^^
2024-07-28T08:20:53.092442013Z   File "/opt/venv/lib/python3.11/site-packages/urllib3/connectionpool.py", line 843, in urlopen
2024-07-28T08:20:53.092455692Z     retries = retries.increment(
2024-07-28T08:20:53.092468621Z               ^^^^^^^^^^^^^^^^^^
2024-07-28T08:20:53.092481330Z   File "/opt/venv/lib/python3.11/site-packages/urllib3/util/retry.py", line 519, in increment
2024-07-28T08:20:53.092494918Z     raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
2024-07-28T08:20:53.092508235Z     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-07-28T08:20:53.092538154Z urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='moonraker', port=80): Max retries exceeded with url: /machine/device_power/device?device=caselight&action=on (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f93fff2c690>: Failed to establish a new connection: [Errno 111] Connection refused'))

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2024-08-25:
@yeahimongithub pls check `development` branch
this should be fixed)

yeahimongithub, 2024-09-22:
I just tried to verify by pulling the latest development-container.
Sadly I receive an error on a different place now...
```
2024-09-22T08:08:44.030747452Z   File "/opt/moonraker-telegram-bot/bot/camera.py", line 141, in __init__
2024-09-22T08:08:44.030750440Z     if cv2.ocl.haveOpenCL():
2024-09-22T08:08:44.030753367Z        ^^^^^^^
2024-09-22T08:08:44.030756297Z AttributeError: 'NoneType' object has no attribute 'ocl'
2024-09-22T08:08:44.030759202Z Stack (most recent call last):
2024-09-22T08:08:44.030762077Z   File "/opt/moonraker-telegram-bot/bot/main.py", line 68, in handle_exception
2024-09-22T08:08:44.030765059Z     logger.error(
2024-09-22T08:09:44.946405914Z 2024-09-22 08:09:44,945 - __main__ - ERROR - main.py:68 - Uncaught exception
2024-09-22T08:09:44.946428072Z Traceback (most recent call last):
2024-09-22T08:09:44.946431797Z   File "/opt/moonraker-telegram-bot/bot/main.py", line 1212, in <module>
2024-09-22T08:09:44.946435258Z     else FFmpegCamera(configWrap, klippy, rotatingHandler) if configWrap.camera.cam_type == "ffmpeg" else Camera(configWrap, klippy, rotatingHandler)
2024-09-22T08:09:44.946438400Z 
```

i suspect that the import is running into an error:
``` py
try:
    import cv2  # type: ignore
except ImportError:
    cv2 = None
```

nlef, 2024-09-25:
@yeahimongithub images with tag `developmet` should work. pls update to latest and check)

yeahimongithub, 2024-09-25:
Yes, everything works now.
Thanks for your work, i really appreciate it!

closed 2024-09-25 by yeahimongithub