nlef/moonraker-telegram-bot #322: Creality K1 max support vi guilouz installer script

balthus, 2024-08-03:
### What would you like to see added?

Hello,
I wonder why such wonderful telegram bot is not available for Creality K1 running Klipper.
I recently acquired a K1 max and wanted to install this plug-in.
Unfortunately, it doesn't seem straightforward. Bash is not existent, systemctl is a mocked script...
I just wonder if it would be possible to port this plug-in to guilouz installer script (https://guilouz.github.io/Creality-Helper-Script-Wiki/helper-script/helper-script-installation/)

Do you know if someone tried to port this to rooted creality K1 running Klipper?
Are you aware of some issue to try to port/run your code on a rooted K1 with Klipper?

There is a similar telegram plugin but it seems to comes with octoprint which is a third party cloud app... I don't want to rely/depend on external service, so running a telegram bot directly on the printer is a must.

I used an ender3 with octopi and telegram addon, so I really wish to find a similar functionality with the K1.

Thanks for any help/hint/advices

nlef, 2024-08-25:
Hello,
The main problem is cross-compilation of venv for the bot, because on the creative k1 there is a motherboard with the architecture mips32p2, for which there are no assembled packages.
If someone helps with the implementation of the C job for assembling venv, I will gladly add this feature)

balthus, 2024-09-15:
Ok, Thanks for your answer. I understand more now why there is no direct telegram plugin on the K1.
I'm not able to do cross-compilation but with your answer, it's useless for me to try to manually install/replicate the setup on my K1.
I hope it will be possible one day because it's a pity to have to rely on other cloud provider to get some telegram notification.
Thanks again :)

divelgrom, 2024-11-12:
> Hello, The main problem is cross-compilation of venv for the bot, because on the creative k1 there is a motherboard with the architecture mips32p2, for which there are no assembled packages. If someone helps with the implementation of the C job for assembling venv, I will gladly add this feature)

Hi. doesn't `pip3 install virtualenv` solve the problem?

nlef, 2024-11-17:
> Hi. doesn't `pip3 install virtualenv` solve the problem? 

if you have the desire and the opportunity, check the possibility of installing the bot from the development branch on your creality

Also some deps like rust compiler may be needed)