nlef/moonraker-telegram-bot #326: [Bug]:  ModuleNotFoundError: No module named 'cv2'

RNGIllSkillz, 2024-09-08:
### What happened and in what context?

installed with kiauh on armbian iot. moonraker-telegram-bot will not start with ModuleNotFoundError: No module named 'cv2' error. python3-opencv is installed on the system

### If needed, paste the relevant bot log contents here.

Sep 09 01:54:17 orangepi5pro systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Sep 09 01:54:17 orangepi5pro systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE
Sep 09 01:54:17 orangepi5pro python[22647]: ModuleNotFoundError: No module named 'cv2'
Sep 09 01:54:17 orangepi5pro python[22647]:     ^^^^^^^^^^
Sep 09 01:54:17 orangepi5pro python[22647]:     import cv2  # type: ignore
Sep 09 01:54:17 orangepi5pro python[22647]:   File "/home/illskillz/moonraker-telegram-bot/bot/camera.py", line 16, in <module>
Sep 09 01:54:17 orangepi5pro python[22647]:     from camera import Camera
Sep 09 01:54:17 orangepi5pro python[22647]:   File "/home/illskillz/moonraker-telegram-bot/bot/notifications.py", line 14, in <module>
Sep 09 01:54:17 orangepi5pro python[22647]:     from notifications import Notifier
Sep 09 01:54:17 orangepi5pro python[22647]:   File "/home/illskillz/moonraker-telegram-bot/bot/websocket_helper.py", line 12, in <module>
Sep 09 01:54:17 orangepi5pro python[22647]:     from websocket_helper import WebSocketHelper
Sep 09 01:54:17 orangepi5pro python[22647]:   File "/home/illskillz/moonraker-telegram-bot/bot/main.py", line 41, in <module>
Sep 09 01:54:17 orangepi5pro python[22647]: Traceback (most recent call last):
Sep 09 01:54:16 orangepi5pro systemd[1]: Started moonraker-telegram-bot.service - Moonraker Telegram Bot SV1 NST%.
Sep 09 01:54:16 orangepi5pro systemd[1]: Stopped moonraker-telegram-bot.service - Moonraker Telegram Bot SV1 NST%.
Sep 09 01:54:16 orangepi5pro systemd[1]: moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 84.
Sep 09 01:54:06 orangepi5pro systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Sep 09 01:54:06 orangepi5pro systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

snelweg, 2024-09-24:
same here, happend all of a sudden

`moonraker-telegram-bot.service: Scheduled restart job, restart counter is at 2.
Sep 24 19:48:14 ender3 systemd[1]: moonraker-telegram-bot.service: Consumed 1.160s CPU time.
Sep 24 19:48:14 ender3 systemd[1]: moonraker-telegram-bot.service: Failed with result 'exit-code'.
Sep 24 19:48:14 ender3 systemd[1]: moonraker-telegram-bot.service: Main process exited, code=exited, status=1/FAILURE
Sep 24 19:48:14 ender3 python[3420]: ModuleNotFoundError: No module named 'cv2'
Sep 24 19:48:14 ender3 python[3420]:     import cv2  # type: ignore
Sep 24 19:48:14 ender3 python[3420]:   File "/home/ton/moonraker-telegram-bot/bot/camera.py", line 16, in <module>
Sep 24 19:48:14 ender3 python[3420]:     from camera import Camera
Sep 24 19:48:14 ender3 python[3420]:   File "/home/ton/moonraker-telegram-bot/bot/notifications.py", line 14, in <module>
Sep 24 19:48:14 ender3 python[3420]:     from notifications import Notifier
Sep 24 19:48:14 ender3 python[3420]:   File "/home/ton/moonraker-telegram-bot/bot/websocket_helper.py", line 12, in <module>
Sep 24 19:48:14 ender3 python[3420]:     from websocket_helper import WebSocketHelper
Sep 24 19:48:14 ender3 python[3420]:   File "/home/ton/moonraker-telegram-bot/bot/main.py", line 41, in <module>
Sep 24 19:48:14 ender3 python[3420]: Traceback (most recent call last):
Sep 24 19:48:13 ender3 systemd[1]: Started Moonraker Telegram Bot SV1 NST%.
Sep 24 19:48:13 ender3 systemd[1]: moonraker-telegram-bot.service: Consumed 1.505s CPU time.
Sep 24 19:48:13 ender3 systemd[1]: Stopped Moonraker Telegram Bot SV1 NST%.
`

`Using cached tornado-6.1-cp39-cp39-manylinux2014_aarch64.whl (427 kB)
Using cached charset_normalizer-3.3.2-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (138 kB)
Installing collected packages: wsaccel, rel, pytz, websocket-client, urllib3, ujson, tzlocal, tornado, six, PySocks, Pillow, idna, emoji, charset-normalizer, certifi, cachetools, requests, APScheduler, python-telegram-bot
Successfully installed APScheduler-3.6.3 Pillow-9.4.0 PySocks-1.7.1 cachetools-4.2.2 certifi-2024.8.30 charset-normalizer-3.3.2 emoji-2.2.0 idna-3.10 python-telegram-bot-13.15 pytz-2024.2 rel-0.4.8 requests-2.28.2 six-1.16.0 tornado-6.1 tzlocal-2.1 ujson-5.7.0 urllib3-1.26.20 websocket-client-1.5.1 wsaccel-0.6.4
[OK] Installing Python requirements successful!

###### Creating new Moonraker Telegram Bot Instance ...
[OK] Service file created: /etc/systemd/system/moonraker-telegram-bot.service
[OK] Env file created: /home/ton/printer_data/systemd/moonraker-telegram-bot.env

###### Enable moonraker-telegram-bot.service ...
[OK] OK!

###### Start moonraker-telegram-bot.service ...
[OK] OK!

###### Add section '[update_manager moonraker-telegram-bot]' to '/home/ton/printer_data/config/moonraker.conf' ...
[ERROR] Error during installation of Moonraker Telegram Bot:
'str' object has no attribute 'keys'
`

snelweg, 2024-09-25:
installed `opencv-contrib-python` no change
When I run a python interperter and do a 
`import cv2`
there is no error and the module is found.

nlef, 2024-09-25:
@snelweg `python3-opencv` package is required. After installing the package, recreate bot venv or reinstall bot using a script `scripts/install.sh`
@RNGIllSkillz what errors appear when trying to install a package `python3-opencv`?

RNGIllSkillz, 2024-09-25:
> @RNGIllSkillz what errors appear when trying to install a package `python3-opencv`?

No errors. python3-opencv is already the newest version (4.6.0+dfsg-12).

snelweg, 2024-09-26:
Found the solution (at least for me). If I use the Kiauh   v6.0.0-alpha.4 install then the bot does not work. When I do a manual install, as per instruction here, then it works!

RNGIllSkillz, 2024-09-26:
I can confirm. installing it manually seems to fix the issue.

sshaikh, 2024-09-29:
Had the same issue via KIAUH. Have raised it there (with an alternative fix is `install.sh` doesn't work).

nlef, 2024-12-06:
@sshaikh @snelweg @RNGIllSkillz
in kiauh version 6 there were some fixes regarding the bot installation. Is there a problem with installing the bot dependencies now?

closed 2025-01-04 by nlef