nlef/moonraker-telegram-bot #334: [Bug]: Bot cant reposnse and have empty log

jc-skurman, 2024-10-28:
### What happened and in what context?

After installation via KIAUH, the bot does not respond to telegrams and also has an empty config (I added token and chat_id)

[image]

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

jc-skurman, 2024-10-29:
in system journalctl status I get error:
```
Oct 29 20:58:17 Voron systemd[1]: Started Moonraker Telegram Bot SV1.
Oct 29 20:58:26 Voron python[1411124]: 2024-10-29 20:58:26,973 - __main__ - ERROR - Uncaught exception
Oct 29 20:58:26 Voron python[1411124]: Traceback (most recent call last):
Oct 29 20:58:26 Voron python[1411124]:   File "/home/biqu/moonraker-telegram-bot/bot/main.py", line 1080, in <module>
Oct 29 20:58:26 Voron python[1411124]:     conf.remove_option("bot", "bot_token")
Oct 29 20:58:26 Voron python[1411124]:   File "/usr/lib/python3.9/configparser.py", line 943, in remove_option
Oct 29 20:58:26 Voron python[1411124]:     raise NoSectionError(section) from None
Oct 29 20:58:26 Voron python[1411124]: configparser.NoSectionError: No section: 'bot'
Oct 29 20:58:26 Voron python[1411124]: Stack (most recent call last):
Oct 29 20:58:26 Voron python[1411124]:   File "/home/biqu/moonraker-telegram-bot/bot/main.py", line 64, in handle_exception
Oct 29 20:58:26 Voron python[1411124]:     logger.error(

```

vvboyko92, 2024-11-01:
I have the same issue. Have you sorted it out somehow?

jc-skurman, 2024-11-01:
Unfortunately, not yet.  I have an idea to add a log of the location of the telegram config file in the main.py file and see where it expects the configuration file.  But while my printer is being repaired, I can't do it.

jc-skurman, 2024-11-01:
.

jc-skurman, 2024-11-01:
closed by mistake

closed 2024-11-01 by jc-skurman

vvboyko92, 2024-11-01:
> Unfortunately, not yet. I have an idea to add a log of the location of the telegram config file in the main.py file and see where it expects the configuration file. But while my printer is being repaired, I can't do it.

check install.sh
```
[Service]
Type=simple
User=${CURRENT_USER}
ExecStart=${MOONRAKER_BOT_ENV}/bin/python ${MOONRAKER_BOT_DIR}/bot/main.py -c ${MOONRAKER_BOT_CONF}/telegram.conf -l ${MOONRAKER_BOT_LOG}
Restart=always
RestartSec=5
EOF
```

```
cat /etc/systemd/system/moonraker-telegram-bot.service
[Unit]
Description=Moonraker Telegram Bot SV1
Documentation=https://github.com/nlef/moonraker-telegram-bot/wiki
After=network-online.target

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=klipper
WorkingDirectory=/home/klipper/moonraker-telegram-bot
EnvironmentFile=/home/klipper/printer_data/systemd/moonraker-telegram-bot.env
ExecStart=/home/klipper/moonraker-telegram-bot-env/bin/python $TELEGRAM_BOT_ARGS
Restart=always
RestartSec=10
```

```
cat /home/klipper/printer_data/systemd/moonraker-telegram-bot.env
TELEGRAM_BOT_ARGS="/home/klipper/moonraker-telegram-bot/bot/main.py -c /home/klipper/printer_data/config/printer.cfg -l /home/klipper/printer_data/logs/telegr
```

in my case it takes printer.cfg and from directory where the file is missing
I will try to change the path to the the right config file and let you know

jc-skurman, 2024-11-01:
Maybe need include telegram.cfg in printer.cfg?

vvboyko92, 2024-11-01:
> Maybe need include telegram.cfg in printer.cfg?

Resolved! modified the env file of the service
/home/klipper/printer_data/systemd/moonraker-telegram-bot.env
and restarted the services

`TELEGRAM_BOT_ARGS="/home/klipper/moonraker-telegram-bot/bot/main.py -c /home/klipper/printer_data/config/telegram.conf -l /home/klipper/printer_data/logs/telegram.log"
`

jc-skurman, 2024-11-01:
can you try not change env file  but include telegram conf file and test the same?

vvboyko92, 2024-11-01:
> can you try not change env file but include telegram conf file and test the same?

doesn't work, it says that [bot] is the wrong config parameter