nlef/moonraker-telegram-bot #335: [Bug]: if the bot is hosted on another server, it loses connection and does not restore it after turning on the printer

Tombraider2006, 2024-10-29:
### What happened and in what context?

Bot online, no moonraker connection!
 Connection failed. 
Failing...

after restoring the printer power, you need to give the command /bot_restart otherwise nothing happens

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2025-01-06:
@burylo @Tombraider2006  
fixed in [319ad1b](https://github.com/nlef/moonraker-telegram-bot/commit/319ad1b13bfa9e1f4502c4f9eb14001210c778d6)
please test the changes

nlef, 2025-01-19:
fixed in 2.1

closed 2025-01-19 by nlef