nlef/moonraker-telegram-bot #339: [Bug]: power toggle won't work

cruzer619, 2024-12-09:
### What happened and in what context?

After upgrading to latest version of telegram-bot when printer is off but pi is still on.  Power toggle no longer works.  I've tried to uninstall and reinstall. and same result.  Also when first starting the printer.  And assuming the service starts.  Printer online comes on every 30's.  After I stop and restart service it will stop.

### If needed, paste the relevant bot log contents here.

*******************************************************************
Current Moonraker telegram bot config
[bot]
server = localhost
power_device = Ender5
light_device = LED1

[progress_notification]
percent = 10

*******************************************************************
2024-12-09 01:38:35,438 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:36,455 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:37,467 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:38:37,472 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:38,487 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:39,501 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:38:39,505 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:40,519 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:41,536 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:38:41,540 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:42,562 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:43,599 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:44,603 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:38:44,622 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/info'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:45,645 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:38:46,267 - klippy - ERROR - klippy.py:362 - Server error '503 Klippy Host not connected' for url 'http://localhost/printer/objects/list'
For more information check: https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503
2024-12-09 01:38:47,435 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
trunk
2024-12-09 01:48:15,435 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:48:17,435 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:48:19,437 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:48:21,437 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
2024-12-09 01:48:22,413 - __main__ - ERROR - main.py:986 - Exception while handling an update:
Traceback (most recent call last):
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpx/_transports/default.py", line 101, in map_httpcore_exceptions
    yield
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpx/_transports/default.py", line 394, in handle_async_request
    resp = await self._pool.handle_async_request(req)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 256, in handle_async_request
    raise exc from None
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_async/connection_pool.py", line 236, in handle_async_request
    response = await connection.handle_async_request(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_async/connection.py", line 101, in handle_async_request
    raise exc
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_async/connection.py", line 78, in handle_async_request
    stream = await self._connect(request)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_async/connection.py", line 124, in _connect
    stream = await self._network_backend.connect_tcp(**kwargs)
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_backends/auto.py", line 31, in connect_tcp
    return await self._backend.connect_tcp(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/httpcore/_backends/anyio.py", line 122, in connect_tcp
    stream._raw_socket.setsockopt(*option)  # type: ignore[attr-defined] # pragma: no cover
[... 41 lines cut ...]
    return await self._post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/_bot.py", line 617, in _post
    return await self._do_post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/ext/_extbot.py", line 351, in _do_post
    return await super()._do_post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/_bot.py", line 646, in _do_post
    result = await request.post(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/request/_baserequest.py", line 202, in post
    result = await self._request_wrapper(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/request/_baserequest.py", line 334, in _request_wrapper
    code, payload = await self.do_request(
  File "/home/pi/moonraker-telegram-bot-env/lib/python3.9/site-packages/telegram/request/_httpxrequest.py", line 310, in do_request
    raise TimedOut from err
telegram.error.TimedOut: Timed out
2024-12-09 01:48:23,440 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}
trunk
2024-12-09 01:53:49,434 - websocket_helper - WARNING - websocket_helper.py:312 - Error received from websocket: {'code': 503, 'message': 'Klippy Host not connected'}

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

cruzer619, 2024-12-12:
ok was able to fix the printer online status repeating every 30 secs.  Referencing #326, I used kiauh alpha 6 to reinstall.  I manually installed via git.  So I'm good there, but "/power" function doesn't work unless I'm at my home computer to turn it on.
[image] 
on 1.5.0 I was able to power on printer when it was in the off state.

cruzer619, 2024-12-12:
ok rolled back to 1.5.0 and now able to power on printer when its off.  so using RP 4 with relay to turn on/off printer mainboard via moonraker.

nlef, 2024-12-31:
@cruzer619 
Most likely you are using power devices that control klipper output_pin. In the latest versions of moonraker, when the klipper is in shutdown/error state, switching such power devices is impossible.
The API for controlling power devices has not changed in moonraker, the logic for working with it has not changed in the bot.
For convenience, I added the output of an error message in the reply message when controlling power devices from the bot in [add8df3](https://github.com/nlef/moonraker-telegram-bot/commit/add8df380d32a946842da9e03319ef31ac5d18bf)
It would be cool if you could test and indicate the errors that occur when working with power devices for you.

cruzer619, 2025-01-02:
[image]
@nlef yes I am using GPIO pins to toggle relay on/off.  ok just got home from out of state.  Will give this a try and send error test message.

cruzer619, 2025-01-02:
[telegram.log](https://github.com/user-attachments/files/18289132/telegram.log)
Here is the latest telegram log

nlef, 2025-01-02:
@cruzer619 
judging by the attached log file, you are using a bot from the master branch. Unfortunately, you won't be able to test the commit I specified above. Could you add `debug: true` to the section `bot` in the bot config file, restart the bot, switch the power device and attach a new bot log?

cruzer619, 2025-01-02:
[image]
my bad.. I thought I grabbed development branch.. saw in notes you added "power device error"...  so from this screen.  I thought if I clicked on the green code button, and copied the url from there would git the develpment branch.  or is there another way of doing it?

cruzer619, 2025-01-02:
[telegram.log](https://github.com/user-attachments/files/18289993/telegram.log)
Here is new log with bot => debug: true

nlef, 2025-01-02:
```
cd ~/moonraker-telegram-bot
git checkout development
```

also you could add `primary_branch: development` into moonraker config update_manager section for the bot. This will give the ability to update the bot from the dev branch from the web interface

cruzer619, 2025-01-02:
ok thanks will update and try again... but will need some sleep..  will try again in about 5 hours..  thanks.

cruzer619, 2025-01-02:
[telegram.log](https://github.com/user-attachments/files/18293054/telegram.log)
ok here is new log.  I updated to development branch.  turned off debug mode.  I did like 5 power toggles and after about 3 min "5 power on dialogs" showed up.  hit power on and printer turned on.  but when I pressed power toggle again. about 5 times.  no diaglog to turn off printer.  so its not just to power it on.  seems to be lack of some communication.  as I tried status as well.  and no dialog either with a pic.

cruzer619, 2025-01-03:
FYI I've started to print my ERCF v2 parts.  Won't be able to test 'til the following day.

nlef, 2025-01-04:
@cruzer619 
a logical bug with blocking in power devices was found and fixes 
please update to the latest version in the development branch and test

cruzer619, 2025-01-04:
good morning.. ( well for me )  will update now and test

nlef, 2025-01-05:
judging by the logs, the bot cannot connect to the moonraker. I added processing and outputting connection errors to the moonraker in the status command respond.
The welcome message in version 2 appears a little later. The bot tries to connect to the moonraker for up to 1 minute and in case of a successful connection or expiration of the specified time, it sends a message about the connection or with the error text

cruzer619, 2025-01-05:
roger will update repo and try again.. its jus strange light switch acknowledges right away every time...

cruzer619, 2025-01-05:
[telegram.log](https://github.com/user-attachments/files/18311726/telegram.log)
Here is the log and here is snap shot of my sequence.
[image]

cruzer619, 2025-01-06:
ok updated to latest dev repo and your 15 sec max timeout working.  I only have to do a power switch one time.  Will test other switches but have a feeling its all fixed except for status and IP camera .

nlef, 2025-01-06:
In version 2 camera settings have been changed and parameter `type` was added with the default value `mjpeg`.
If your remote camera is not streaming mjpeg, you have to set `type`:`opencv` in camera settings.
Pls take a look at the release notes of version 2 and wiki

cruzer619, 2025-01-06:
@nlef correct I was looking at that.  At first was working but after reboot it failed..  was using:

[camera]
host = http://192.168.1.110:8081/
type = ffmpeg
host_snapshot = http://192.168.1.110/picture/1/frame/

will try again.

cruzer619, 2025-01-06:
After more testing and reboots all is fixed with power toggle and status on dev branch.  closing ticket.

closed 2025-01-06 by cruzer619

nlef, 2025-01-06:
@cruzer619  Thank you very much for testing and help!
I fixed the typo in error message and changed schedulling logic for greeting message, so you should receive "Bot online" messages after bot starts.

cruzer619, 2025-01-06:
@nlef sure its my pleasure to help out.. and after udating to latest dev repo.. welcome message shows up.. and quick testing all is working.  sweat!!!!!