nlef/moonraker-telegram-bot #340: [Bug]: Bot won't start on Qidi Q1 Pro

danb35, 2024-12-12:
### What happened and in what context?

I'm attempting to install and run this bot on my new Qidi Q1 Pro printer.  I've used it successfully on other printers, but they were running generic Klipper; this one is running a "dirty" version of 0.10 on a MKS Pi board running Armbian Buster.  Reports are that trying to run current, clean Klipper will brick the printer, so I haven't attempted to do that.  The software installation they provide includes Kiauh, and using that I was able to update both Fluidd and Moonraker to current, and it appeared to install the bot, but the bot is never able to start.

Looking at the system logs, it's failed to start due to a variety of missing python modules, many of which I've managed to address by `sudo pip3 install modulename`.  But I'm currently stuck on `ModuleNotFoundError: No module named 'websockets.asyncio'`, and I can't come up with a way to resolve that.

I'm not attaching `telegram.log`, because it doesn't exist--presumably because the bot never starts.  Journal output is attached.  Any suggestions would be appreciated.

### If needed, paste the relevant bot journal contents here.

[telegram-journal.txt](https://github.com/user-attachments/files/18117151/telegram-journal.txt)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

nlef, 2024-12-12:
@danb35 hi!
you can try to install the bot with a script in the bot repo `scripts/install.sh`
all dependencies with versions are specified in the file `scripts/requirements.txt`
dependencies need to be installed in the bot venv

unfortunately support for non-mainline clipper, strange OS builds will not be provided

Megastrong1, 2024-12-21:
Привет! Установи Open Q1 и сможешь поставить все что ты хочешь! Единственное что не будет работать это экран. А в остальном этот образ переплюнет оригинальный образ Q1.

closed 2025-01-04 by nlef