nlef/moonraker-telegram-bot #346: [Bug]: "Klippy changed state to ..." message spam - only enable selected notifications

GoTVm, 2024-12-26:
### What happened and in what context?

I have two Klipper printers, each with their own Pi and Klipper/Moonraker install. The older one does not send any message when Klipper is started/encounters an error, whereas the new one sends a message for each Klippy state change and also when the printer is online.

I would like to only receive the messages I specified in the notifier section, as below (moonraker.conf)

```
[notifier print_end]
url: tgram://************
events: complete
body: Print complete '{event_args[1].filename}'
attach: http://localhost/webcam/?action=snapshot

[notifier print_error]
url: tgram://************
events: error
body: Print error '{event_args[1].filename}'
attach: http://localhost/webcam/?action=snapshot
```

and disable every other notification.

### If needed, paste the relevant bot log contents here.

[telegram.log](https://github.com/user-attachments/files/18254010/telegram.log)

### What branch does this occur on?

master

### Reports only on latest master or development branches!

- [X] I have updated to latest development/master version before submitting the bug

aka13-404, 2024-12-26:
What does our bot have to do with your moonraker notifier section?
The bot sends an update on every moonraker status change, and this is intended behaviour, which has been this way for a very long time.

GoTVm, 2024-12-26:
My older printer probably doesn't have Moonraker Telegram Bot configured, just installed. I might've mixed up the notifier section with Moonraker Telegram Bot.

How long has this been default behaviour? I configured Telegram notifications on my older printer around February of last year. I don't have my printer at hand right now, so I am unable to check what packages are installed.

aka13-404, 2025-01-19:
Pretty much since the creation of this project, and this is not likely going to change in the foreseeable future.

closed 2025-01-19 by aka13-404