nlef/moonraker-telegram-bot #357: Disable Time-lapse functionality, and resources or compute.

PhilBaz, 2025-02-01:
### What would you like to see added?

Hello,

I wish there was an easy way to verify that the time-lapse function was not capturing any images or using any resources. 

I'm not interested in time-lapse, yet there doesn't seem to be a way to disable it completely? Or maybe by not using it it is not consuming any resources? I don't know, and its not clear or discussed in the docs.

aka13-404, 2025-02-01:
It takes all of my/my colleagues effort to reply in a non-condescending matter here. 
If you did not mean this as a provocation, there is nothing to verify, and there will be nothing to verify. Don't add features you don't plan on using to your config. How much resources get used by what you can get by running a profiler or inspecting the sourcecode.

closed 2025-02-01 by aka13-404

PhilBaz, 2025-02-01:
Not a provocation...I'm not a computer guy. Its not clear if the time-lapse is totally disactivated by default.

aka13-404, 2025-02-01:
Ah, alright. It's completely off when not present in the config, and does not take any pictures.