nlef/moonraker-telegram-bot #372: [Bug]: Failed to get status

khthe8th, 2025-07-24:
### What happened and in what context?

Installed moonraker telegram bot using KIAUH alpha v6. Requesting /status shows picture of webcam but message of "Failed to get status: '[Errno -2] Name or service not known. 

[image]

Telegram.conf is as shown:

[image]

I am able to access http://192.168.1.181/printer/objects/query?webhooks&print_stats&display_status from a computer successfully, which should be the same URL the bot tries to access?

Telegram log:
[telegram.log](https://github.com/user-attachments/files/21401066/telegram.log)

Output of sudo journalctl -r -u moonraker-telegram-bot

[image]

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [x] I have updated to latest development/master version before submitting the bug

sandrohtc, 2025-07-25:
i had the same problem. i fix it by removing http:// from server

khthe8th, 2025-07-25:
> i had the same problem. i fix it by removing http:// from server

Thanks, that solved it!

closed 2025-07-25 by khthe8th