nlef/moonraker-telegram-bot #375: [Bug]: MULTIPLE ISTANCE NOT WORKING  moonraker-telegram-bot.service

IvanAmatiDotCom, 2025-09-05:
### What happened and in what context?

I installed Klipper on a Speeder Pad with multiple instances (2).
I installed Telegram Bot via Kiau, but it doesn't work. When I follow the wiki and it tells me to run "sudo systemctl restart moonraker-telegram-bot," the terminal says "moonraker-telegram-bot.service failed."

It's not a hardware problem because Moonraker notifications work. I tried installing Octoprint with its plugin, and it works perfectly, but I can't get it to work.
I followed the entire guide, modified the .conf file with token and chat ID, but the bot remains silent. No commands are issued, as if they aren't communicating. Can anyone help me? I've been struggling for 5 days trying to figure it out.

I have 2 instances, printer_1_data and printer_2_data, as per the default.

### What branch does this occur on?

development

### Reports only on latest master or development branches!

- [x] I have updated to latest development/master version before submitting the bug

aka13-404, 2025-09-05:
We do not support any cut down chinesium klipper installs. 
This is not a bug, contact whoever set up the system/manufacturer.

closed 2025-09-05 by aka13-404

IvanAmatiDotCom, 2025-09-06:
> We do not support any cut down chinesium klipper installs. This is not a bug, contact whoever set up the system/manufacturer.

I would like to point out that a clean version of Klipper was installed on this hardware by deleting the old dependencies and in any case I have the exact same problem on a Raspberry Pi3B too, so?